juribeparada/MMDVM_HS #153: Dual Hat USB bootloader installation fails

StefanSchwan, 2023-01-22:
I have a dual hat device with USB (Aursinc Brand from Amazon). 

pistar-findmodem identifies it as 
`Modem Data: MMDVM_HS_Dual_Hat-v1.5.2 20201108 14.7456MHz dual ADF7021 FW by CA6JAU GitID #89daa2000480067340000084E533443`

I have soldered `JP1` and start to device jumpered to `BOOT`
 
it is working without any problems over GPIO on a PI4. I would like to enable the bootloader and use it on as a USB device on Ubuntu 22.10.

On my Ubuntu box, I use a generic ST-LINK device and I am trying to compile and flash a firmware from current master. 

```
#define MMDVM_HS_DUAL_HAT_REV10
#define ENABLE_ADF7021
#define ADF7021_14_7456
#define AD7021_GAIN_AUTO
#define STM32_USB_HOST
#define ENABLE_SCAN_MODE
```
I run 

```
make clean
make bl
sudo make stlink-bl 
```

(output see blow, claims to be `jolly good`)

The device is not recognoized when I attach it via USB (dmesg silent). I have tried it on a PI and a Macbook Pro also.  

When I compile using `#define STM32_USART1_HOST`, I can `pistar-findmodem` the modem with the right commit hash for current master on my PI - but it does not receive anything. 

I have only got it work again using `pistar-mmdvmhshatflash hs_dual_hat`. This will revert it to `v1.5.2`

I have tried compiling and flashing `v1.5.2` myself, this fails with an error 
```
unknown coreid, not sure what flash loader to use, aborting!: 2ba01477
```

On master, I have `ST_FLASH_ON_PATH` with 

❯ /usr/bin/st-flash --version
v1.7.0

How can I get this to work?

```
❯ sudo make stlink-bl
/usr/bin/st-flash --flash=128k write ./STM32F10X_Lib/utils/bootloader/generic_boot20_pc13_long_rst.bin 0x8000000
st-flash 1.7.0
2023-01-21T21:42:26 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file ./STM32F10X_Lib/utils/bootloader/generic_boot20_pc13_long_rst.bin md5 checksum: 4b2f1eeb1a8aaa192d6255e94a341c, stlink checksum: 0x000900fc
2023-01-21T21:42:26 INFO common.c: Attempting to write 7180 (0x1c0c) bytes to stm32 address: 134217728 (0x8000000)
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08000000 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08000400 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08000800 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08001000 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08001400 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08001800 erased
2023-01-21T21:42:26 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-01-21T21:42:26 INFO common.c: Finished erasing 8 pages of 1024 (0x400) bytes
2023-01-21T21:42:26 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-01-21T21:42:26 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-01-21T21:42:26 INFO flash_loader.c: Clear DFSR
  8/  8 pages written
2023-01-21T21:42:26 INFO common.c: Starting verification of write complete
2023-01-21T21:42:26 INFO common.c: Flash written and verified! jolly good!
sleep 3
/usr/bin/st-flash --flash=128k write bin/mmdvm_f1bl.bin 0x8002000
st-flash 1.7.0
2023-01-21T21:42:29 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file bin/mmdvm_f1bl.bin md5 checksum: d3e618b2dede3a40ad16b126d987357, stlink checksum: 0x004085dd
2023-01-21T21:42:29 INFO common.c: Attempting to write 51356 (0xc89c) bytes to stm32 address: 134225920 (0x8002000)
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08002000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08002400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08002800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08002c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08003000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08003400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08003800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08003c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08004000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08004400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08004800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08004c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08005000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08005400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08005800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08005c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08006000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08006400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08006800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08006c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08007000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08007400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08007800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08007c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08008000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08008400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08008800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08008c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08009000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08009400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08009800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x08009c00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x0800a000 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x0800a400 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x0800a800 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x0800ac00 erased
2023-01-21T21:42:29 INFO common.c: Flash page at addr: 0x0800b000 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800b400 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800b800 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800bc00 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800c000 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800c400 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800c800 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800cc00 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800d000 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800d400 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800d800 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800dc00 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800e000 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800e400 erased
2023-01-21T21:42:30 INFO common.c: Flash page at addr: 0x0800e800 erased
2023-01-21T21:42:30 INFO common.c: Finished erasing 51 pages of 1024 (0x400) bytes
2023-01-21T21:42:30 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-01-21T21:42:30 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-01-21T21:42:30 INFO flash_loader.c: Clear DFSR
2023-01-21T21:42:30 INFO flash_loader.c: Clear CFSR
2023-01-21T21:42:30 INFO flash_loader.c: Clear HFSR
2023-01-21T21:42:30 INFO common.c: Go to Thumb mode
 51/ 51 pages written
2023-01-21T21:42:31 INFO common.c: Starting verification of write complete
2023-01-21T21:42:31 INFO common.c: Flash written and verified! jolly good!
```

ndoolin, 2023-03-21:
I'm having a similar issue.
I posted here:
https://github.com/phl0/MMDVM_HS_Dual_Hat/issues/10

Have you been able to resolve?

Thanks!

-Nick

ndoolin, 2023-03-22:
Update:  I was able to get the Bootloader written to the MMDVM Board.  The issue seemed to be related to a possibly cloned chip that was not recognized in the older versions of st-flash.  I downloaded the latest stlink tools and compiled.  I additionally uninstalled the old stlink-tools package which was an older version.  Re-running st-flash still resulted in the same unknown coreid issue.  The uninstall did not unlink the libstlink.so.1 and was still pointing to the older version.  I re-linked to the new version of libstlink and the bootloader install finally worked.  However, I'm still not able to communicate with the board over USB.  I've tried both with JP1 soldered and not soldered.  I have not done anything with the BOOT 0 or BOOT 1 pins.  The device will report as Maple 003 in Windows but no matter which driver I've attempted to assign(libusb, WinUSB, devser) via zadig BlueDV cannot communicate with the board.  I've tried flashing precompiled firmware that indicates they have USB support(https://github.com/juribeparada/MMDVM_HS/releases/download/v1.5.2/generic_duplex_usb_fw.bin) as well as compiling my own mmdvm_f1bl.bin and flashing that.  I've also attemtped to flash these with the STM32Programmer with the same results.  

Anyone else have ideas on this?

Thanks!

-Nick