juribeparada/MMDVM_HS #156: Failed to init device error while manualy installing the firmware

py2oc, 2023-03-31:
Hello.

I have got an error trying to manualy installing the MMDVM firmware on pi-star. The board is an hs dual hat.
The firmware compiling runs without error but when trying to install, results on "Failed to init device, timeout" message.
What am I missing?

---------------------------------------------------------------------------------------------------------------
pi-star@pi-star(rw):MMDVM_HS$ sudo pistar-watchdog.service stop
Killing pistar-watchdog PID 1288
pi-star@pi-star(rw):MMDVM_HS$ sudo systemctl stop mmdvmhost.timer
pi-star@pi-star(rw):MMDVM_HS$ sudo systemctl stop mmdvmhost.service
pi-star@pi-star(rw):MMDVM_HS$ sudo make mmdvm_hs_dual_hat
/usr/local/bin/stm32flash -v -w bin/mmdvm_f1.bin -g 0x0 -R -i 20,-21,21:-20,21 /dev/ttyAMA0
stm32flash 0.7

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Size         : 57032
Interface serial_posix: 57600 8E1

GPIO sequence start
 setting gpio 20 to 1... OK
 delay 100000 us
 setting gpio 21 to 0... OK
 delay 100000 us
 setting gpio 21 to 1... OK
GPIO sequence end

Failed to init device, timeout.

GPIO sequence start
 setting gpio 20 to 0... OK
 delay 100000 us
 setting gpio 21 to 1... OK
GPIO sequence end

make: *** [Makefile:484: zumspot-pi] Error 1
---------------------------------------------------------------------------------------------------------

73 de PY2OC, Luiz

W0CHP, 2023-03-31:
This is not a bug; Very common when the STM32 chip is locked and/or not writable. Either you need to enable the jumper or solder it yourself. Search on how to do that.

py2oc, 2023-03-31:
Thanks W0CHP. The board had a bad solder on jumper pins.
Now I could install the firmware but got a new issue. Pistar receives the "Received a NAK to the SET_CONFIG" error message and can't initialize the modem.

W0CHP, 2023-04-01:
> Thanks W0CHP. The board had a bad solder on jumper pins. Now I could install the firmware but got a new issue. Pistar receives the "Received a NAK to the SET_CONFIG" error message and can't initialize the modem.

Not a bug, and nothing to do with this firmware...

You are likely trying to run FW v1.6.0 on Pistar which has an older MMDVMhost, thus, MMDVMhost can't understand some of the new FW commands. This is not a bug ; it's a known incompatibility. Pi-Star ships with MMDVMhost that supports up to v1.5.2 of the HAT FW.

Three options:

1. Downgrade the FW to v1.5.2 (which pistar's older MMDVMHost supports)
2. Compile and install MMDVMhost and MMDVMCal to the latest so it can communicate with the FW (however you risk this breaking during PiSar updates)
3. Upgrade your Pi-Star instance to a "well-known fork", which already ships all of the updated binaries that support v1.6.0 FW.

Please close this issue, as it has nothing to do with the MMDVM_HS firmware and is not a bug.

py2oc, 2023-04-04:
Ok. Thanks for help.

closed 2023-04-04 by py2oc