juribeparada/MMDVM_HS #161: FW 1.6.1 !?

labdvm, 2024-02-11:
Hi, when will available new code for FW 1.6.1 ? Thanks.

g4klx, 2024-02-11:
You need to get it from the G4KLX GitHub for now.

msiqx9, 2024-05-15:
Within Pi-Star I just updated the firmware to v1.6.1

f5vmr, 2025-06-07:
I've recently taken ownership of a generic hotspot and uploaded this version, but I suspect that possibly that YSF hasn't been enabled in the Config.h. I am viewing YSF Data emanating from the network in the YSF Log with debug and full logging enabled, with a room to which I have connected, but none of the data is being displayed within the dashboard. None of the transmissions I am attempting to the hotspot including the calibrate function are being acknowledged by the hotspot, while perfectly acceptable with DMR. The repeater in the same room is quite happy with YSF, and I'm not likely to update the repeater at this time (It's a ZUM-Radio 0.9 and quite capable as it is.) Any comments please?